mercedes-benz/odxtools #4: improve readability of the output of the `list` subcommand

andlaus, 2021-11-12:
currently, the `list` subcommand produces output like this:
```
$ odxtools list --conformant -a examples/somersault.pdx
BASE-VARIANT 'somersault' (Receive ID: 0x7b, Send ID: 0x1c8)
 num services: 7, num DOPs: 7, num communication parameters: 11.
The services of the BASE-VARIANT 'somersault' are:
 do_backward_flips <ID: somersault.service.do_backward_flips>
  Message format of a request:
           7     6     5     4     3     2     1     0
        +-----+-----+-----+-----+-----+-----+-----+-----+
      0 | sid (8 bits)                                  |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      1 | backward_soberness_check (8 bits)             |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      2 | num_flips (8 bits)                            |
        +-----+-----+-----+-----+-----+-----+-----+-----+
   Parameter(short_name='sid', type='CODED-CONST', semantic=None, byte_position=0, bit_length=8, coded_value='0xbb')
   Parameter(short_name='backward_soberness_check', type='VALUE', semantic=None, byte_position=1, bit_length=8, dop_ref='somersault.DOP.soberness_check')
    DataObjectProperty('soberness_check', category='IDENTICAL', internal_type='A_UINT32', physical_type='A_UINT32')
   Parameter(short_name='num_flips', type='VALUE', semantic=None, byte_position=2, bit_length=8, dop_ref='somersault.DOP.num_flips')
    DataObjectProperty('num_flips', category='IDENTICAL', internal_type='A_UINT32', physical_type='A_UINT32')
  Number of positive responses: 1
  Message format of a positive response:
           7     6     5     4     3     2     1     0
        +-----+-----+-----+-----+-----+-----+-----+-----+
      0 | sid (8 bits)                                  |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      1 | num_flips_done (8 bits)                       |
        +-----+-----+-----+-----+-----+-----+-----+-----+
   Parameter(short_name='sid', type='CODED-CONST', semantic=None, byte_position=0, bit_length=8, coded_value='0xfb')
```

I think it would be more user friendly if this was something like

```
Variants:
 somersault:
  Type: 'BASE-VARIANT'
  Receive ID: 0x7b
  Send ID: 0x1c8
  Services:
   do_backward_flips
    Request parameters:
     sid:
      Type: 'CODED-CONST'
      Semantic: None
      Byte position: 0
      Bit length: 8
      Coded value: '0xbb'
   backward_soberness_check:
    Type: 'VALUE'
    Semantic: None
    Byte position=1
    Bit length: 8
    DOP: 'somersault.DOP.soberness_check'
  Message layout:
           7     6     5     4     3     2     1     0
        +-----+-----+-----+-----+-----+-----+-----+-----+
      0 | sid (8 bits)                                  |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      1 | backward_soberness_check (8 bits)             |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      2 | num_flips (8 bits)                            |
        +-----+-----+-----+-----+-----+-----+-----+-----+
  Positive responses:
   [...]
[...]
  DOPs:
    num_flips:
     Category: 'IDENTICAL'
     Internal type: 'A_UINT32'
     Physical type: 'A_UINT32'
[...]
```

kakoeh, 2024-01-05:
Output format was changed with #250 
The displayed output is now:
```
$ odxtools list -a examples/somersault.pdx
$ odxtools list $BASE_DIR/odxtools/examples/somersault.pdx --variants somersault_lazy --services do_forward_flips --params
Overview of diagnostic layers:
    | Name                | Variant Type | Num. of Services | Num. of DOPs | Num. of comparams
----+---------------------+--------------+------------------+--------------+-------------------
  0 | somersault          | BASE-VARIANT |                7 |           10 |                10
  1 | somersault_lazy     | ECU-VARIANT  |                5 |           10 |                10
  2 | somersault_assiduous| ECU-VARIANT  |                8 |           10 |                10

Diagnostic layer: 'somersault'
 Variant Type: BASE-VARIANT
 Description: Base variant of the somersault ECU & cetera

The global negative responses of 'somersault' are:
 gnr_name_1

The services of 'somersault' are:
 do_backward_flips <ID: OdxLinkId('somersault.service.do_backward_flips')>
  Service description: [...]
  Enabled-Audiences: [...]

  Request Properties:
   Request Name: do_forward_flips
   Byte-Array: ---      Hex-String: 0x---
   Service Parameters: [sid, forward_soberness_check, num_flips]

    | Name                    | Byte Pos. | Bit Length | Param. Type | Data Type | Value | Value Desc. | Linked DOP
----+-------------------------+-----------+------------+-------------+-----------+-------+-------------+-----------------
  0 | sid                     |         0 |          8 | CODED-CONST | A_UINT32  | 0xBA  | coded value |
  1 | forward_soberness_check |         1 |          8 | VALUE       | A_UINT32  |       |             | soberness_check
  2 | num_flips               |         2 |          8 | VALUE       | A_UINT32  |       |             | num_flips

  Message format of a request:
           7     6     5     4     3     2     1     0
        +-----+-----+-----+-----+-----+-----+-----+-----+
      0 | sid(8 bits)                                   |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      1 | forward_soberness_check(8 bits)               |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      2 | num_flips(8 bits)                             |
        +-----+-----+-----+-----+-----+-----+-----+-----+
   
  Positive Response Properties:
   Number of Positive Responses: 1
   Positive Responses: [grudging_forward]
   Service Parameters: [sid, num_flips_done]

    | Name           | Byte Pos. | Bit Length | Parameter Type         | Data Type | Value | Value Desc. | Linked DOP
----+----------------+-----------+------------+------------------------+-----------+-------+-------------+-------------
  0 | sid            |         0 |          8 | CODED-CONST            | A_UINT32  | 0xFA  | coded value |
  1 | num_flips_done |         1 |          8 | MATCHING-REQUEST-PARAM |           |       |             |

  Message format of a positive response:
           7     6     5     4     3     2     1     0
        +-----+-----+-----+-----+-----+-----+-----+-----+
      0 | sid(8 bits)                                   |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      1 | num_flips_done(8 bits)                        |
        +-----+-----+-----+-----+-----+-----+-----+-----+
   
  Negative Response Properties:
   Number of Negative Responses: 1
   Negative Responses: [flips_not_done]
   Service Parameters: [sid, rq_sid, reason, flips_successfully_done]

    | Name                    |Byte Pos. | Bit Length | Parameter Type         | Data Type | Value     | Value Desc. | Linked DOP
----+-------------------------+----------+------------+------------------------+-----------+-----------+-------------+-------------
  0 | sid                     |        0 |          8 | CODED-CONST            | A_UINT32  | 0x7F      | coded value |
  1 | rq_sid                  |        1 |          8 | MATCHING-REQUEST-PARAM |           |           |             |
  2 | reason                  |        2 |          8 | NRC-CONST              | A_UINT32  | [0, 1, 2] | coded value |
  3 | flips_successfully_done |        3 |          8 | VALUE                  | A_UINT32  |           |             | num_flips

  Number of negative responses: 1
  Message format of a negative response:
           7     6     5     4     3     2     1     0
        +-----+-----+-----+-----+-----+-----+-----+-----+
      0 | sid(8 bits)                                   |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      1 | rq_sid(8 bits)                                |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      2 | reason(8 bits)                                |
        +-----+-----+-----+-----+-----+-----+-----+-----+
      3 | flips_successfully_done(8 bits)               |
        +-----+-----+-----+-----+-----+-----+-----+-----+
   
```

kakoeh, 2024-01-05:
More ideas for improvement: 
- change from tabulate to rich tables

Katja Köhler &lt;katja.koehler@mercedes-benz.com&gt;, on behalf of [MBition GmbH](https://mbition.io/).
[Provider Information](https://github.com/mercedes-benz/foss/blob/master/PROVIDER_INFORMATION.md)

andlaus, 2024-10-30:
I suppose that this issue can be closed for now. PRs improving the `list` output are still appreciated, though...

closed 2024-10-30 by andlaus

Haris-Ali007, 2024-11-03:
Hello @kakoeh I have create a PR to improve the documentation. #359