mercedes-benz/odxtools #5: code generators for c++ and other languages

andlaus, 2021-11-12:
to be able to implement diagnostic services with low-cost hardware (i.e.,  microcontrollers for embedded systems or machines exhibiting less than 28 cores in the cloud world), it is necessary to use a compiled language, most commonly C++ (or C, but IMO even for microcontrollers restricting oneself to this is pretty masochistic). The way odxtools could help on this front is to generate the c++ boiler plate code necssarry for encoding and decoding UDS telegrams which needs to be filled by the developer wishing to implement a diagnostic service. This is basically the same approach as `cantools generate_c_source`, but I'd warmly recommend to use a proper template engine such as [jinja](https://jinja.palletsprojects.com/en/3.0.x/) for this.

On the c++ side, this could look similar to:

```c++
#include <FunkyECU/SomersaultBaseService.hh>

namespace FunkyECU {
    class SomersaultService : public SomersaultBaseService<SomersaultService>
    {
    public:
        SomersaultService(IsoTpConnection& conn)
          : SomersaultBaseService(conn)
          , num_done_(0)
        {}

        void request_received(const SomersaultRequestParams& rq)
        {
           num_done_ += rq.param_NumLoops;

            if (num_done_ < 10) {
                SomersaultPositiveResponse resp;
                resp.param_NumLeft = 10 - num_done_;
                this->sendResponse(resp);
            }
            else {
                // i am dizzy now!
                this->sendResponse(SomersaultNegativeResponse);
            }
        }

    private:
        int num_done_;
    };
}

int main()
{
   IsoTpConnection conn(SomersaultService::receive_id(), SomersaultService::send_id());
   FunkyECU::SommersaultService service(conn);
   while (true) {
       IsotpTelegram telegram = conn.recv()
       if (service.applies(telegram))
           service.handle(telegram);
   }
   return 0;
}
```

illegal-instruction-co, 2023-07-31:
When I compare the purpose of this repository and the request in the issue, I believe that this should not be the responsibility of this project. Implementation is a specific task that individuals/companies have to undertake.

Odxtools is a data business assistant designed to assist with data analysis and research. In my opinion, it is not appropriate to expect a full implementation within odxtools.

When considering low-performance embedded systems, instead of testing the code directly on the device, it would be more practical to try it in the vehicle using a laptop that can emulate the device or mirror the bus. As a result, code written in C or C++ should function the same way on two different devices, producing identical results.

Mehmet Berkay Karatas

andlaus, 2023-07-31:
Note that this is not a "request" in the usual sense, it is more an idea for a feature that could be useful.

Implementing a code generator for compiled languages is currently not a priority for me personally, but I will not turn down pull requests for this (at least not because I would consider it out-of-scope). If you mean shipping implementations based on top of the generated API definitions for concrete ECUs, I agree with your statement...

driftregion, 2023-12-20:
Hello. Please consider using my project https://github.com/driftregion/iso14229 . It implements the session layer of UDS as well as a few application-layer correctness checks (e.g. length, format, etc.). odxtools could potentially be used to generate repetitive application-layer code such as RDBI and WDBI.