mercedes-benz/odxtools #6: extend unittests

andlaus, 2021-11-12:
Currently, test coverage of the code is quite low. This should be improved.

kayoub5, 2022-11-10:
@andlaus in order to know what's covered and what's not you will need to setup something like https://codecov.io/
I could do the codecov configuration once a it is activated on the mercedes-benz GitHub organization

andlaus, 2022-11-11:
ok, I checked what codecov wants, but since they request read/write permissions for the *entire* mercedes-benz github organization (i.e., public and private repos, contributor permissions to all repos, etc.), it is not gonna fly. I've checked what [coveralls](https://coveralls.io/) (used by [cantools](https://github.com/cantools/cantools)) requests, and they only want write access to all PR statuses of the organization, which *might* be doable, but I need to check...

andlaus, 2026-10-29:
IMO, this issue has become a dangerous invitation for AI slop contributions, as AI is very good in producing unit tests without the author gaining a deeper understanding of the code or finding loose ends of the code. Let's close the issue.

closed 2026-10-29 by andlaus