mercedes-benz/odxtools #35: add comparison tool

andlaus, 2022-03-16:
having a tool which compares two ECU versions and/or complete databases would be nice. something along the lines of:

```
# compare ECUs
odxtools compare PDX_FILE.pdx somersault_lazy somersault_assiduous

# compare complete files
odxtools compare PDX_FILE_OLD.pdx PDX_FILE_NEW.pdx
```

KarloGaloz, 2023-09-16:
Hi Andreas,

I'm checking out the project rn and I'm interested in adding the feature.
You can assign me to this issue if you like.

andlaus, 2023-09-16:
From my side that not really required (just open a PR once there's something to show), but I'm happy if you want to work on this :)

KarloGaloz, 2023-09-16:
Awesome - will do :)

kakoeh, 2024-01-05:
PR #249 introduced  `compare` as new command line utility.
The `compare` subcommand can be used to compare databases (pdx-files) and diagnostic layers with each other. All diagnostic services as well as its parameters of specified databases and variants are compared with each other and changes are displayed.

Examples:
  Comparison of two ecu versions:
    `odxtools compare ./path/to/database.pdx -v variant1 variant2`
  Comparison of two database versions:
    `odxtools compare ./path/to/database.pdx -db ./path/to/old-database.pdx`

Katja Köhler &lt;katja.koehler@mercedes-benz.com&gt;, on behalf of [MBition GmbH](https://mbition.io/).
[Provider Information](https://github.com/mercedes-benz/foss/blob/master/PROVIDER_INFORMATION.md)

kakoeh, 2024-01-05:
Ideas for improvement of comparison tool: #252

andlaus, 2024-01-08:
this issue has been completed via #249. closing...

closed 2024-01-08 by andlaus