mercedes-benz/odxtools #40: cannot load pdx file in version 1.2.2

minkeetan, 2022-04-25:
Hi,

I tried to load pdx with the latest version but it failed. It was working in previous version.
The following is the log.

  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\odxtypes.py", line 64, in from_string
    return _ODX_TYPE_PARSER[self.value](value)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'xml.etree.ElementTree.Element'

---

One more thing I would like to ask is there any example on how to send the UDS service to the ECU after loading the pdx? For example like Read Data by Identifier service.

andlaus, 2022-04-25:
for the exception: can you provide a complete traceback?

sending data: yes, it's possible. once you've encoded your request, you can send it to the ECU using basically any python ISO TP package. See e.g. the [isotp_send](https://github.com/mercedes-benz/odxtools/blob/main/examples/isotp_send.py) example for details.

minkeetan, 2022-04-25:
@andlaus ,

Here is the complete traceback:
Traceback (most recent call last):
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\Scripts\odxtools.exe\__main__.py", line 7, in <module>
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\__init__.py", line 107, in _main
    _main.start_cli()
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\cli\main.py", line 51, in start_cli
    list.run(args)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\cli\list.py", line 131, in run
    odxdb = _parser_utils.load_file(args)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\cli\_parser_utils.py", line 26, in load_file
    odxdb = _load_file(db_file_name,
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\load_file.py", line 9, in load_file
    return load_pdx_file(file_name, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\load_pdx_file.py", line 11, in load_pdx_file
    container = Database(pdx_zip=u, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\database.py", line 45, in __init__
    tmp = [
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\database.py", line 46, in <listcomp>
    read_diag_layer_container_from_odx(dlc_el, enable_candela_workarounds=enable_candela_workarounds) \
    base_variants = [read_diag_layer_from_odx(dl_element, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\diaglayer.py", line 586, in read_diag_layer_from_odx
    diag_data_dictionary_spec = read_diag_data_dictionary_spec_from_odx(
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\diagdatadictionaryspec.py", line 98, in read_diag_data_dictionary_spec_from_odx
    data_object_props = [read_data_object_property_from_odx(dop_element)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\diagdatadictionaryspec.py", line 98, in <listcomp>
    data_object_props = [read_data_object_property_from_odx(dop_element)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\dataobjectproperty.py", line 318, in read_data_object_property_from_odx
    compu_method = read_compu_method_from_odx(et_element.find(
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\compumethods\readcompumethod.py", line 138, in read_compu_method_from_odx
    compu_inverse_value = internal_type.from_string(v)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools\odxtypes.py", line 64, in from_string
    return _ODX_TYPE_PARSER[self.value](value)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'xml.etree.ElementTree.Element'

andlaus, 2022-04-25:
> Thanks for the answer. How about sending UDS request over Ethernet? DOIP?

in principle it works the same, i.e., like with CAN, you can encode the payload using odxtools but the actual send operation is out of scope, i.e. you need a external python package for that...

andlaus, 2022-04-25:
#43 should fix this. @minkeetan can you check?

minkeetan, 2022-04-25:
It fix this one, but then it goes to another issue

Traceback (most recent call last):
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\Scripts\odxtools-script.py", line 33, in <module>
    sys.exit(load_entry_point('odxtools==1.2.2', 'console_scripts', 'odxtools')())
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\__init__.py", line 107, in _main
    _main.start_cli()
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\cli\main.py", line 51, in start_cli
    list.run(args)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\cli\list.py", line 131, in run
    odxdb = _parser_utils.load_file(args)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\cli\_parser_utils.py", line 26, in load_file
    odxdb = _load_file(db_file_name,
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\load_file.py", line 9, in load_file
    return load_pdx_file(file_name, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\load_pdx_file.py", line 11, in load_pdx_file
    container = Database(pdx_zip=u, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\database.py", line 45, in __init__
    tmp = [
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\database.py", line 46, in <listcomp>
    read_diag_layer_container_from_odx(dlc_el, enable_candela_workarounds=enable_candela_workarounds) \
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\diaglayer.py", line 699, in read_diag_layer_container_from_odx
    base_variants = [read_diag_layer_from_odx(dl_element, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\diaglayer.py", line 699, in <listcomp>
    base_variants = [read_diag_layer_from_odx(dl_element, enable_candela_workarounds=enable_candela_workarounds)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\diaglayer.py", line 586, in read_diag_layer_from_odx
    diag_data_dictionary_spec = read_diag_data_dictionary_spec_from_odx(
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\diagdatadictionaryspec.py", line 101, in read_diag_data_dictionary_spec_from_odx
    structures = [read_structure_from_odx(structure_element)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\diagdatadictionaryspec.py", line 101, in <listcomp>
    structures = [read_structure_from_odx(structure_element)
  File "C:\Users\tnm1pg1\AppData\Local\Programs\Python\Python39\lib\site-packages\odxtools-1.2.2-py3.9.egg\odxtools\structures.py", line 394, in read_structure_from_odx
    long_name = et_element.find("LONG-NAME").text
AttributeError: 'NoneType' object has no attribute 'text'

---
FYI,
I replaced line 394 at structures.py with 
long_name = et_element.find("LONG-NAME").text if et_element.find("LONG-NAME") is not None else short_name
and it works.

andlaus, 2022-04-25:
this should already be fixed in the latest master. release coming...

minkeetan, 2022-04-26:
@andlaus , tested with version 1.2.3, it works fine. Thanks.

closed 2022-04-26 by minkeetan