mercedes-benz/odxtools #41: Is there any way to encode UDS service?

minkeetan, 2022-04-25:
I knew there is a way to decode UDS service from a raw data, for instance :

raw_data = b"\x22\x10\x31"
decoded_message = ecu.decode(raw_data)
print(f"decoded message: {decoded_message}")

But is there anyway to encode UDS service?
so that it can be used to send the message to the ECU.

andlaus, 2022-04-25:
I suppose you need to have an ODX file which describes the generic UDS services for this. If decoding using a specific ECU's odx file fails, you can fall back to the generic one. (If you have such a file, we'd be interested to include it directly into odxtools.) alternatively, you can have a look at third-party packages like python-uds. I have to confess that I don't have any experience with this, though.

minkeetan, 2022-04-25:
Yes, I have that odx / pdx file. 
What I dont know is how to extract that particular service out from odx and send it to ECU over ethernet.

lets say the diagnostic service that I wish to send is a self_check_routine. After loading the pdx file, Is it possible for me to encode self_check_routine to get the raw_data? 
so that it can be used to send the message to the ECU via ethernet network.

andlaus, 2022-04-25:
as said above: the actual sending of the raw data needs to be done externally. determining the payload data to be done can be done using something like this:
```
import odxtools
db = odxtools.load_file("uds.pdx")
ecu = db.ecus.standard_uds
raw_data = ecu.services.read_dtcs()
# send raw_data via DoIP
```

andlaus, 2022-04-25:
(that assumes that your ODX file for the UDS standard is called `uds.pdx`, the ecu specified therein is called `standard_uds` and the service to read the diagnostic trouble codes is called `read_dtcs`.)

minkeetan, 2022-04-26:
Thanks @andlaus . Got it.

closed 2022-04-26 by minkeetan