mercedes-benz/odxtools #83: CAN send/receive ids are null

smileprem, 2022-10-18:
Hi team - We were using the commit `74163b455f070f8fd925955bb33911bfcd83a092` which was a few weeks old. Recently upgraded to the latest version at `3b33829178dc854e8a578ff66be7459137017391`

With the new version of the library, the can send/receive ids appear as `None`. This was working with the previous version.

A sample test program. 
```
import odxtools

database = odxtools.load_file("./Sibros_TCU_V1.0.1.pdx", enable_candela_workarounds=True)
for dlc in database.diag_layer_containers:
    for ecu_variant in dlc.base_variants:
        print("physical_request_id = ", hex(ecu_variant.get_receive_id()))
        print("physical_response_id = ", hex(ecu_variant.get_send_id()))
```
Error:
```
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    print("physical_request_id = ", hex(ecu_variant.get_receive_id()))
TypeError: 'NoneType' object cannot be interpreted as an integer

```

Gitlab does not allow me to upload the PDX files. Probably you can test with the PDX files that you have.
[image]

**Question:** Are these APIs deprecated?. Should I be looking at a different function?. Please advise.

kayoub5, 2022-10-19:
@smileprem you can put the PDX in a ZIP and upload it

smileprem, 2022-10-19:
That's brilliant. Why didn't I think of that?. Here is the PDX file that you can test with.

[Sibros_TCU_V1.0.1.pdx.zip](https://github.com/mercedes-benz/odxtools/files/9822735/Sibros_TCU_V1.0.1.pdx.zip)

Also, I put up an quick MR for this. https://github.com/mercedes-benz/odxtools/pull/84

smileprem, 2022-11-07:
Hi Team - The latest commit from main branch (cdb078019e41c84f4763d15d44166635b86259e4) fixed the issues we observed with the earlier version. Thanks for fixing the errors. I'll close this ticket.

closed 2022-11-07 by smileprem