mercedes-benz/odxtools #152: Does it support ODX 2.2.0?

zzpbuaa, 2023-07-04:
When I try to load a pdx file. it throw the error:

  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\load_file.py", line 9, in load_file
    return load_pdx_file(file_name)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\load_pdx_file.py", line 11, in load_pdx_file
    container = Database(pdx_zip=u)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\database.py", line 86, in __init__
    self.refresh()
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\database.py", line 118, in refresh
    dlc._finalize_init(self._odxlinks)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\diaglayercontainer.py", line 162, in _finalize_init
    dl._finalize_init(odxlinks)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\diaglayer.py", line 215, in _finalize_init
    self.diag_layer_raw._resolve_snrefs(self)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\diaglayerraw.py", line 285, in _resolve_snrefs
    obj._resolve_snrefs(diag_layer)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\diagdatadictionaryspec.py", line 169, in _resolve_snrefs
    obj._resolve_snrefs(diag_layer)
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\endofpdufield.py", line 160, in _resolve_snrefs
    self._structure = dops[self.structure_snref]
  File "D:\PythonEnvs\py38\lib\site-packages\odxtools\nameditemlist.py", line 89, in __getitem__
    return self._typed_dict[key]
KeyError: 'STRUCT_Service19_DTCExtRecord'
python-BaseException

I'm wondering if it's because my pdx file is in ODX 2.2.0 format?

kayoub5, 2023-07-04:
@andlaus this looks to be simular to #149 but for endofpdufield, the logic needs to be double checked for other cases as well.

andlaus, 2023-07-04:
> Does it support ODX 2.2.0

2.2.0 is actually the targeted version. (I only have the spec for that...) That said, we accept patches fixing 2.0 support if they don't break 2.2, but it is difficult to make any commitments about this without the ability to test...

andlaus, 2023-07-07:
fixed by #155. closing.

closed 2023-07-07 by andlaus