mercedes-benz/odxtools #156: ODXTools : Admin data access in Individual Variant in the ECU

jayeshmanani, 2023-07-05:
can not access the admin data for the individual variant inside the one ECU, It seems it is not available now, as able to access the admin data in the diag layer container for the doc revision for ECU, but not individual variants admin data like doc revisions in one of the variants of the ECU - like for boot variant, base variant etc.

andlaus, 2023-07-06:
are you sure that you use the latest release? Like all diagnostic layers, ECU variants should feature an `admin_data` property from odxtools version 4.0 on:

(cf. https://github.com/mercedes-benz/odxtools/blame/main/odxtools/diaglayer.py#L240-L242, https://github.com/mercedes-benz/odxtools/blob/main/odxtools/diaglayerraw.py#L46, and https://github.com/mercedes-benz/odxtools/blob/main/odxtools/diaglayerraw.py#L91-L94)

jayeshmanani, 2023-07-26:
Yeah Found It

closed 2023-07-26 by jayeshmanani