mercedes-benz/odxtools #157: Can we make a flag to make it support parsing PDX file even validation fail?

chenyin0126, 2023-07-05:
We discussed in a previous [question](https://github.com/mercedes-benz/odxtools/issues/140) about parsing PDX files when validation fails. The validation failure is caused by the keyword "assert", for example, 

`def parse_int(value: str) -> int:
    try:
        return int(value)
    except ValueError:
        v = float(value)
        assert v.is_integer()
        return int(v)`

In this scenario, if the input string is "0.05", the entire process fails. However, we would like to continue parsing the rest of the validated PDX file without letting one type of issue block the entire process. Editing the PDX file is a costly process, and we would prefer not to modify it.

Is it possible to introduce a flag that allows us to control the behavior of the library, enabling it to parse the entire PDX file even if validation fails in a testing environment? For instance, we can toggle the flag to false during testing, allowing the parsing of the PDX file. However, in the production environment, we can set the flag to true to ensure parsing fails if validation fails. The production environment should prioritize safety considerations.

andlaus, 2023-07-06:
the problem with this in general is that removing the asserts outright will lead to a lot of mypy complaints. In your particular case, I would advise to locally remove the incriminating `assert` in `parse_int()`, if you're *really* sure that throwing away all decimal digits is correct...

also, it could be that `parse_int()` was used somewhere where floating point values are allowed. In this case, please open a PR fixing that. (this is the case if all `*.odx-d` files that are contained within the PDX validate against the XSD schema file from the ODX specification package.)

andlaus, 2023-09-05:
since #168 has been merged, there now is a non-strict mode which might help here. Let's close the issue for now. If it turns out that the non-strict mode does not help, feel free to re-open it...

closed 2023-09-05 by andlaus