mercedes-benz/odxtools #160: How to encode a ReadDataByIdentifier request?

zzpbuaa, 2023-07-14:
Hi guys,
    I'm trying to encode a ReadDataByIdentifier request. It should provide a DID, how can I do this? I don't have a ecu in my pdx file. So I tried to use function like db.diag_layers['VCU_Android_GAS'].services['ReadDataByIdentifier'].request.encode(). But I don't know how to pass the DID to the function **encode()**
   I can get DID list from db.diag_layers['VCU_Android_GAS'].diag_data_dictionary_spec.tables['TABLE_Service22_DataID'] for ReadDataByIdentifier because its did number is 22.

andlaus, 2023-07-14:
if you call `encode()` for the request, it should complain about missing mandatory parameters. Alternatively you can use `db.diag_layers.VCU_Android_GAS.services.ReadDataByIdentifier.request.print_free_parameters_info()`. passing parameters works by simply passing them to the `encode()` method, e.g.
```python
rq.encode(foo_param=123, bar="HELLO_TEXTTABLE_KEY")
```

note that `encode()` expects physical quantities, not unscaled ones...

zzpbuaa, 2023-07-17:
Hi @andlaus, could you do me a favor to modify the layer name VC***S to **myLayer** in your comment? It's my bad. Because this word may be a keyword related to my customer's information. I'm not sure if I'll be in trouble for unintentionally leaking this info or not. If you can help to modify it I'll appreciate it very much.

Back to the question, when I use 'print_free_parameters...', I got the below output:
```
DataIds : <optional> list({
  DataId : <unhandled parameter type>
})
```

It needs a DataId parameter, it's a composite type, not a simple type like int or string. So I don't know which type I should pass in.

andlaus, 2023-07-17:
> Back to the question, when I use 'print_free_parameters...', I got the below output:

this means that the kind of parameter used by `DataId` is not yet implemented for en-/decoding. Amongst others, this could e.g. be `STATIC-FIELD`, `DYNAMIC-FIELD`, or `TABLE-STRUCT`. (I'm currently working on support for table parameters, but it will be a week or two until this is hits the `main` branch...) You can find out what it is via
```python
dl = db.diag_layers.myLayer
rq = dl.services.ReadDataByIdentifier.request
param = rq.DataId
print(param.parameter_type)
```

zzpbuaa, 2023-07-18:
The parameter_type is "VALUE". Then this is a DYNAMIC-FIELD or TABLE-STRUCT right?

andlaus, 2023-07-18:
table key parameters in themselves do not make too much sense, as they only specify which row ought to apply to a subsequent table struct parameter (i.e., they are the multiplexor IDs of a multiplexer mechanism). As mentioned above, I'm currently working on table en- and decoding, so stay tuned for a few days/weeks...

andlaus, 2023-08-09:
this seems to be solved? if it isn't, feel free to re-open the issue...

closed 2023-08-09 by andlaus