mercedes-benz/odxtools #198: Program Code Syntax

jochen-zimmer, 2023-09-05:
Hi, 
is there any particular reason for exactly those "types"?
https://github.com/mercedes-benz/odxtools/blob/d58a298456ddf40a214cb9b79e1aba711c726f31/odxtools/progcode.py#L14
I could not find any information on valid or invalid types and I get an error on [this line](https://github.com/mercedes-benz/odxtools/blob/d58a298456ddf40a214cb9b79e1aba711c726f31/odxtools/progcode.py#L39), which was working with an older version, because my odx file is using the type "txt" here. Not sure if the ODX is not following the norm or if 'txt' would need to be added as type here. Any advice?
Additionally, I think here, the string should be converted to all-upper case, just to prevent some false errors (in my case, the type was lower case, so even if there was 'TXT' available as type, is would still fail):
https://github.com/mercedes-benz/odxtools/blob/d58a298456ddf40a214cb9b79e1aba711c726f31/odxtools/progcode.py#L36

Thanks a lot in advance for your feedback and also for the tool of course :)

andlaus, 2023-09-05:
thanks for the catch. This used to be a `Literal` which is an enum type that is only checked during type checking but not at runtime. In #172 I converted most users of `Literal` to `Enum` classes precisely to introduce checks at run time. What I didn't check is if these users of `Literal` were specified by the XSD or if they were just internal debugging aids. For `ProgCodeSyntax` it was the latter (the XSD says the `SYNTAX` sub-tag of `SINGLE-ECU-JOB` is a string...)

andlaus, 2023-09-05:
#199 fixes that issue (I'll make a point release once it has been merged)