mercedes-benz/odxtools #252: improve comparison tool

kakoeh, 2024-01-05:
Currently the `compare` command accepts a list of pdx-files and a list of variants as input arguments.
It would be nice to have an optional input argument where a folder with multiple pdx files can be specified.

Idea: provide folder with multiple pdx files as argument
- load all pdx files in folder, sort them alphabetically, compare databases pairwaise

When multiple pdx files are analyzed, a summary of all changes might be helpful
Idea:
- calculate metrics (number of added services, number of changed services, number of removed services, total number of services per ecu variant, ...)
- display metrics graphically

Katja Köhler &lt;katja.koehler@mercedes-benz.com&gt;, on behalf of [MBition GmbH](https://mbition.io/).
[Provider Information](https://github.com/mercedes-benz/foss/blob/master/PROVIDER_INFORMATION.md)

vinothk-master, 2025-06-01:
Hi @kakoeh

kakoeh, 2025-06-02:
Hi @vinothk-master, 
Sure, we'd appreciate your contribution.

kakoeh, 2025-08-07:
Hi @vinothk-master,
I'd suggest to introduce a new table with change metrics such as the number of added or deleted services. 

The table that is currently generated, provides an overview of selected characteristics (variant type, number of services, number of DOPs, number of communication parameters) of different diagnostic layers within one file. When multiple files are compared, the benefit of this contribution and of a separate table would be to display the identified changes clearly and concisely. 

From my point of view, such table could be structured as follows:

[image]

vinothk-master, 2025-08-27:
> Hi [@vinothk-master](https://github.com/vinothk-master), thank you for your work and time.
> 
> Regarding the table, I wouldn't specify the changes per diagnostic layer. As `Variant` and `Diag Layer` refer to the same object, the number of added, changed or deleted variants can be aggregated per file (database) but not per diagnostic layer. Therefore, I would add up the number of added, deleted and changed variants as well as services for each comparison of two files, e.g.: <img alt="Image" width="897" height="114" src="https://private-user-images.githubusercontent.com/142314987/479049545-af5998af-6206-4ac0-a8bf-67f9787d46a8.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYzMzE5NDYsIm5iZiI6MTc1NjMzMTY0NiwicGF0aCI6Ii8xNDIzMTQ5ODcvNDc5MDQ5NTQ1LWFmNTk5OGFmLTYyMDYtNGFjMC1hOGJmLTY3Zjk3ODdkNDZhOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyN1QyMTU0MDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OGIyZmU3MTQxYmU4OTFjNDRlMWRiMzQ2MzNjMGVkZmUwZmJlY2Y1YjFhNzE2NDAwM2RkN2YzOWJmOGM2ZjJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5rLg6wCkB2u8yZtqN-UfFpJkPc4qggN795kjKbrfhx0">
> 
> However, printing change metrics for each diagnostic layer (if existing) might be instructive as well. In this case, I would print a table such as: <img alt="Image" width="688" height="132" src="https://private-user-images.githubusercontent.com/142314987/479053273-5650d9a5-e390-440d-8e27-8778de2199bb.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYzMzE5NDYsIm5iZiI6MTc1NjMzMTY0NiwicGF0aCI6Ii8xNDIzMTQ5ODcvNDc5MDUzMjczLTU2NTBkOWE1LWUzOTAtNDQwZC04ZTI3LTg3NzhkZTIxOTliYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyN1QyMTU0MDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYWYwMTQ3ZTQ1NTI0NTdlNzI5ZDI1YWViNzQ5YTBiYjZkOWM3ZjU1MDI0YjNlZDEwZjdjNDJlOTUzM2NlNWQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.P3yXddeTfRmSnCuWBPpQKakgczQKR6JBRAwmjeQg8CU">

Hi @kakoeh,

I tried making the changes; however, `rich print` doesn't support merge cells. May I know the way you have implemented the merge cells, please? 

Below is my implementation,
[image]

Thanks
Vinoth Kannan