mercedes-benz/odxtools #287: Import all classes in the toplevel `__init__.py`

andlaus, 2024-04-06:
this would allow to import all classes directly in external code instead of having to know the sub-module where the class of interest is defined. example: With this,
```python
from odxtools.diaglayer import DiagLayer
```
could be replaced by
```python
from odxtools import DiagLayer
```

On the flipside, this increases the effort required for maintenance, because *all* classes provided by the project would need to be imported to `__init__.py`.

vinothk-master, 2025-04-23:
Hi @andlaus

andlaus, 2025-04-25:
I'm not sure anymore if this is a good idea: Whenever a new class is introduced, implementing this issue would require also adapting `__init__.py`. I suspect that this will be forgotten frequently. maybe @kayoub5 can chime in?

andlaus, 2026-10-29:
I came to the conclusion that this not a good idea given that it produces significant maintenance issues and comes with a very limited set of benefits compared to importing odxtools submodules using their full path. Closing.

closed 2026-10-29 by andlaus