mercedes-benz/odxtools #324: Not support TABLE-SNREF?

lz4v4, 2024-07-30:
when i change the table reference type in the example "somersault.pdx" from "<TABLE-REF ID-REF="somersault.table.last_flip_details" />" to "<TABLE-SNREF SHORT-NAME="last_flip_details"/>"

this error appears:
```
  File "/Library/Python/3.8/site-packages/odxtools/parameters/tablekeyparameter.py", line 103, in _resolve_snrefs
    self._table = resolve_snref(self.table_snref, tables, Table)
NameError: name 'Table' is not defined
```

andlaus, 2024-07-31:
thanks for your report. can you test if #325 fixes your issue?

lz4v4, 2024-08-02:
> thanks for your report. can you test if #325 fixes your issue?

I've tested it, and the fix in #325 works perfectly. Thanks a lot!"

commit e77dce00 (2024-08-05): tablekeyparameter: fix short name reference resolution

unfortunately, the `Table` and `TableRow` classes are only available
during type checking (i.e., not during normal execution) because this
would lead to circular imports.

this should fix #324.
closed 2024-08-02 by lz4v4