mercedes-benz/odxtools #331: IdenticalCompuMethod instance in the createanycompumethod.create_any_compu_method_from_et function is missing three required arguments

nada-ben-ali, 2024-08-21:
The [IdenticalCompuMethod instance  in the createanycompumethod.create_any_compu_method_from_et function](https://github.com/mercedes-benz/odxtools/blob/7.3.0/odxtools/compumethods/createanycompumethod.py#L40) is missing three required positional arguments: 'category', 'compu_internal_to_phys', and 'compu_phys_to_internal'.

[image]

[image]

kayoub5, 2024-08-27:
@andlaus FYI

andlaus, 2024-08-27:
okay, this is a bug that must be fixed (I'll do so tomorrow). That said, this code is only excuted if the compu method in question is not known (most likely because it is not implemented. there still are a few of them in odxtools). Can you find out what kind of compu method is featured by your dataset? (i.e. what are the the contents of the `CATEGORY` subtag?)

nada-ben-ali, 2024-08-28:
@andlaus the compu category is **SCALE-RAT-FUNC**.

andlaus, 2024-08-28:
ok, thanks. I suppose I should put some effort into implementing the remaining compu methods. BTW: SCALE-RAT-FUNC is a bit weird IMO because it is not invertible, i.e. encoding a parameter which uses such a compu function is going to be "fun"...

closed 2024-09-02 by nada-ben-ali