mercedes-benz/odxtools #338: Error when parsing Odx-d containing DOPs

Sheero-git, 2024-09-04:
My ODX file surely contains TEXTTABLE compu methods, yet for some reason I am getting an error that  TEXTTABLE compu method must exhibit a COMPU-INTERNAL-TO-PHYS subtag

Any idea what to do?

Here is the log
python3 -m odxtools list -a "odxtools\source.pdx"
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "\__main__.py", line 6, in <module>
    _main()
  File "\__init__.py", line 80, in _main
    _main.start_cli()
  File "\cli\main.py", line 67, in start_cli
    tool.run(args)
  File "\cli\list.py", line 216, in run
    odxdb = _parser_utils.load_file(args)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\cli\_parser_utils.py", line 35, in load_file
    return _load_file(pdx_file_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\odxtools\loadfile.py", line 26, in load_file
    return load_pdx_file(str(file_name))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\odxtools\loadfile.py", line 11, in load_pdx_file
    db.add_pdx_file(str(pdx_file))
  File "\odxtools\database.py", line 56, in add_pdx_file
    self._process_xml_tree(root)
  File "\odxtools\database.py", line 86, in _process_xml_tree
    dlcs.append(DiagLayerContainer.from_et(dlc, []))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "odxtools\diaglayercontainer.py", line 68, in from_et
    ecu_shared_datas = NamedItemList([
                                     ^
  File "\odxtools\diaglayercontainer.py", line 69, in <listcomp>
    EcuSharedData.from_et(dl_element, doc_frags)
  File "odxtools\diaglayers\ecushareddata.py", line 44, in from_et
    ecu_shared_data_raw = EcuSharedDataRaw.from_et(et_element, doc_frags)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "odxtools\diaglayers\ecushareddataraw.py", line 31, in from_et
    kwargs = dataclass_fields_asdict(DiagLayerRaw.from_et(et_element, doc_frags))
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\odxtools\diaglayers\diaglayerraw.py", line 101, in from_et
    diag_data_dictionary_spec = DiagDataDictionarySpec.from_et(ddds_elem, doc_frags)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\odxtools\diagdatadictionaryspec.py", line 80, in from_et
    data_object_props = [
                        ^
  File "odxtools\diagdatadictionaryspec.py", line 81, in <listcomp>
    DataObjectProperty.from_et(dop_element, doc_frags)
  File "\odxtools\dataobjectproperty.py", line 57, in from_et
    compu_method = create_any_compu_method_from_et(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\odxtools\compumethods\createanycompumethod.py", line 31, in create_any_compu_method_from_et
    return TexttableCompuMethod.compu_method_from_et(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\odxtools\compumethods\texttablecompumethod.py", line 31, in compu_method_from_et
    return TexttableCompuMethod(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 8, in __init__
  File "\odxtools\compumethods\texttablecompumethod.py", line 46, in __post_init__
    odxraise("TEXTTABLE compu methods must exhibit a COMPU-INTERNAL-TO-PHYS subtag.")
  File "\odxtools\exceptions.py", line 50, in odxraise
    raise error_type(message)
odxtools.exceptions.OdxError: TEXTTABLE compu methods must exhibit a COMPU-INTERNAL-TO-PHYS subtag.

andlaus, 2024-09-05:
It seems like your dataset contains a texttable that does not specify any mapping from internal to physical values. This is explictly forbidden by the ODX standard (cf. checker rule 64 in section B.2). If you want a "pass through" compu method, change the CATEGORY of the incriminating compu method to  "IDENTICAL".

andlaus, 2024-10-30:
I'll close this issue for now. If you think that some odxtools behaviour should be changed, please add a comment with your proposal and I'll consider implementing it...

closed 2024-10-30 by andlaus

Sheero-git, 2024-10-30:
My pdx file had issues as you previously commented.
Upon resolving them, it worked perfectly.
Thanks!