mercedes-benz/odxtools #342: Is it only for benz?

CCXia1997, 2024-09-15:
It seems error when use pdx of volkswagen.

kayoub5, 2024-09-15:
what's the error you are getting?

andlaus, 2024-09-15:
odxtools is most definitely only for Mercedes cars. That said, I obviously do not have access to diagnostics files made by/for other manufacturers and as @kayoub5 said, we're thus reliant on bug reports/patches from our users. In this case, it is likely that the issue can be fixed if you provided a traceback. also, there is a chance that your files will load if you enable the [non-strict mode](https://github.com/mercedes-benz/odxtools?tab=readme-ov-file#using-the-non-strict-mode) (in this case your mileage will vary, though).

andlaus, 2024-09-15:
odxtools is most definitely not only for Mercedes cars. That said, I obviously do not have access to diagnostics files made by/for other manufacturers and as @kayoub5 said, we're thus reliant on bug reports/patches from our users. In this case, it is likely that the issue can be fixed if you provided a traceback. also, there is a chance that your files will load if you enable the [non-strict mode](https://github.com/mercedes-benz/odxtools?tab=readme-ov-file#using-the-non-strict-mode) (in this case your mileage will vary, though).

Ch4rg3r, 2024-09-17:
Is it possible that the specification is different? It seems that odxtools is based on version 2.2 and 2.0.1 for example is jot fully supported?

CCXia1997, 2024-09-17:
> what's the error you are getting?
Version is 2.0.1 and error when use odxtools.load_odx_d_file() to resolve PROTOCOL file:
**KeyError: "ODXLINK reference OdxLinkRef('ES_XXX') could not be resolved for any of the document fragments [OdxDocFragment(doc_name='ES_XXX', doc_type='LAYER')]"**
ES_XXX is ECU-SHARED-DATA file and PROTOCOL file fragment is:
**<IMPORT-REFS>
     <IMPORT-REF DOCREF="ES_XXX" DOCTYPE="LAYER" ID-REF="ES_XXX" />
</IMPORT-REFS>**

andlaus, 2024-09-19:
> Is it possible that the specification is different? It seems that odxtools is based on version 2.2 and 2.0.1 for example is not fully supported?

yes, that's possible: we target odx 2.2, and odx 2.0 is only supported on a "best effort" basis, i.e., we happily accept patches provided that they do not break support for odx 2.2, but I personally do not have any datasets which use this version, so your mileage may vary.

> KeyError: "ODXLINK reference OdxLinkRef('ES_XXX') could not be resolved  [...]

that does not look like it is related to the ODX version, though. it might be a bug in odxtools (IMPORT-REFs are largely untested), or your datasset might be broken. to test whether the latter is the case, unzip the PDX file (`unzip MY_FILE.pdx`), grep for `ID="ES_XXX"` and check whether the short name of that diagnostic layer matches the DOCREF attribute of the IMPORT-REF.

andlaus, 2024-10-30:
@CCXia1997: can this issue be closed? If not, it would be great if you provided more information...

marcodadamo, 2024-11-03:
explanation is included

andlaus, 2024-11-05:
if you do not provide a (redacted if necessary) traceback or a file exhibiting the issue, we do not stand much chance of fixing this...

andlaus, 2024-11-27:
I'll close this issue as it there is not enough information to determine if there is an issue in odxtools and if there is, what needs to be done to fix it. If you disagree, please provide more details and reopen it...

closed 2024-11-27 by andlaus