mercedes-benz/odxtools #351: MinMaxLengthType throw an error when encoding A_BYTEFIELD parameter

ngoan1608, 2024-10-17:
Hi,

I am facing the issue when encoding the request parameters which contains `A_BYTEFIELD` with `MinMaxLengthType` (`HEX[0..64]`)

When I provided the value as `bytearray` (return from the function `physical_type.base_data_type.from_string()` of that `ValueParameter`) for that param, the below error was thrown:
````
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\diagservice.py", line 254, in encode_request
    return self.request.encode(**kwargs)
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\request.py", line 30, in encode
    self.encode_into_pdu(physical_value=kwargs, encode_state=encode_state)
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\basicstructure.py", line 197, in encode_into_pdu
    param.encode_into_pdu(physical_value=param_phys_value, encode_state=encode_state)
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\parameters\parameter.py", line 139, in encode_into_pdu
    self._encode_positioned_into_pdu(physical_value, encode_state)
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\parameters\valueparameter.py", line 88, in _encode_positioned_into_pdu
    self.dop.encode_into_pdu(physical_value, encode_state=encode_state)
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\dataobjectproperty.py", line 127, in encode_into_pdu
    self.diag_coded_type.encode_into_pdu(internal_value, encode_state)
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\minmaxlengthtype.py", line 78, in encode_into_pdu
    odxraise("MinMaxLengthType is currently only implemented for strings and byte arrays",
  File "C:\Program Files\RobotFramework\python39\lib\site-packages\odxtools\exceptions.py", line 50, in odxraise
    raise error_type(message)
odxtools.exceptions.EncodeError: MinMaxLengthType is currently only implemented for strings and byte arrays
````
I looked into the implementation (`minmaxlengthtype.py`) and observed the below condition (line 77) is not enough:
```
def encode_into_pdu(self, internal_value: AtomicOdxType, encode_state: EncodeState) -> None:
        if not isinstance(internal_value, (bytes, str)):
            odxraise("MinMaxLengthType is currently only implemented for strings and byte arrays",
                     EncodeError)
...
```

The `bytearray` should be added along with `bytes` to check whether the given value is byte arrays.

Thank you,
Ngoan

andlaus, 2024-10-17:
good point. can you open a pull request?

ngoan1608, 2024-10-18:
Hi @andlaus ,

I have opened PR #353 to fix this issue.

Thank you,
Ngoan

andlaus, 2024-10-18:
#353 has been merged. Thanks for the patch, closing.

closed 2024-10-18 by andlaus