mercedes-benz/odxtools #361: Is there a way to visualize the database generated from the pdx

Sheero-git, 2024-11-19:
I'm looking for an api which helps me visualise the data as an excel or sql format

andlaus, 2024-11-21:
I suppose the closest thing to what you want that's shipping out-of-the-box is the `list` tool. That said, since there are basically infinitely many possibilities matching the requirements which you describe above, you probably need to write a script for doing what exactly you need. E.g., if you want to produce a CSV of all diagnostic services but not single-ECU-jobs of all ECU variants in a database, you can use something like this:
```python
import sys
import odxtools

db = odxtools.load_file(sys.argv[1])

print("ECU Name,Service Name")

for ecu in db.ecu_variants:
    for service in ecu.services:
        print(f"{ecu.short_name},{service.short_name}")
```

andlaus, 2024-11-21:
I'm not aware of such a format. odxtools will enable you to easily access all data that's contained in PDX files, but it obviously cannot know what parts of that data you are interested in and how you want that data to be formatted, i.e., you have to write a script doing this (or hire somebody to do it for you)...

closed 2024-11-27 by andlaus