mercedes-benz/odxtools #363: refactor the ´compare´ tool

andlaus, 2024-11-20:
the code `compare` tool is currently a bowl of spaghetti with sub-optimal type annotation support (lots of `type: ignore` comments) and desperately needs a cleanup.

vinothk-master, 2025-02-20:
Hi @andlaus 
Can I contribute to this open issue?
Thanks 
Vinoth Kannan

andlaus, 2025-02-20:
sure, that would be much appreciated. if you're looking for a place to start, a great one would to refactor the [code](https://github.com/mercedes-benz/odxtools/blob/main/odxtools/cli/compare.py) to making use of dataclasses in order to make it easier to understand what data is expected/available where and  to get rid (or at least reduce the number) of `type: ignore` comments...

vinothk-master, 2025-02-20:
Ok @andlaus Thanks for the info, Let me check the code first
Thanks

vinothk-master, 2025-02-21:
Hi @andlaus 

**Status**: I am working on implementing the `@dataclass` as you mentioned and trying to reduce the `type: ignore` comments

Thanks

vinothk-master, 2025-04-23:
Hi @andlaus

andlaus, 2025-04-25:
as I see it, #393 is a great first step, but IIRC there are still a few issues lurking, e.g. using two lists which must be of the same size instead of a single list with objects that contain two attributes (for the removed and added diagcomms of a set of diag layers?) . I have to go over the latest version of the code to come to a definitive conclusion, though.

andlaus, 2026-10-29:
ok, #475 improved this considerably. let's move the compare refactoring discussion over to #252, closing.

closed 2026-10-29 by andlaus