mercedes-benz/odxtools #367: about A_INT32

cmydd, 2024-11-28:
hello, I'm sorry, my English is poor, I hope you can understand my expression
BASE-DATA-TYPE="A_INT32" and BASE-DATA-TYPE="A_UINT32" 
A_INT32  32 bit signed integer (range: -2^31.. 2^31-1) 
A_UNT32 32 bit unsigned integer (range: 0 .. 2^32-1)
so A_INT32  and A_UNT32 are different.
I tested the program
internal_type=DataType.A_INT32,
physical_type=DataType.A_INT32
self.assertEqual(compu_method.convert_physical_to_internal(0x80000001), 7200)
0x80000001 should be a negative number Because it's A_INT32 type and not A_UINT32
I hope you understand what I mean, thank you very much

andlaus, 2024-11-28:
in odxtools, `int` is used as the internal representation of all integer numbers regardless of what the ODX says. Since in python integers are arbitrary length, you will not see any "wrap around" like you would in C. (en- or decoding should work, though it is probably not tested very much...)

cmydd, 2024-11-29:
thinks for your answer. 
When read data from the ECU, it doesn't have a negative number
for example: the read bytes[0x80,0x00,0x00,0x00]
the internal types A_INT32 
(offset + factor * internal_value) / this->denominator
internal_value should be negative number
it‘s read from ecu 0x80000000 A_INT32
not input -0x80000000 A_INT32
it‘s can work.But the result may be wrong
thanks

closed 2024-12-16 by cmydd