mercedes-benz/odxtools #370: when the dop-snref refer to another file.

yangshengquan1026, 2024-12-06:
When the do-snref index in one file is referenced to another file, the program will not run.
[test.zip](https://github.com/user-attachments/files/18038964/test.zip)
[image]

When loading ODX files, each ODX file's content is added to a master library, and in theory, the indexing should not be a problem. However, in practice, the indexing can only be done within the current file, and if the dop-snref is not within the file, the program will not be able to continue executing.
[image]

andlaus, 2024-12-06:
your dataset has an issue: the parameter "DTCFU" which is part of the structure "DTCASVR_STRUCT" defined by ECU-SHARED-DATA layer "SD_UDS_Services" references a DOP called "Uint_IDENTICAL_8Hex" via SNREF which is not defined in that ECU-SHARED-DATA layer. The best way to fix that is to use an ODXLINK reference to the DOP (i.e., specify DOP-REF instead of DOP-SNREF in parameters)

yangshengquan1026, 2024-12-06:
andlaus，谢谢你的快速回复！是的，你说的是对的，我的dop-snref如果在本文件内定义的查询是正确的，但是如果在另外一个文件定义则是查询不到的，这个是用softing的工具生成test.pdx.softing自己的工具解析是没有问题的。我用咱们这个odxtools就出现这个情况，在旧版本上我试着修改过，可以查询到。咱们这个odxtools工具的原理是把所有的文件的字段往一个共同的字典里边存储吧？比如所有的diagservice放到一个字典里边。所有dop放在一个字典里边。我对python不太熟悉，我在试着理解原理然后往C/C++或者rust方向转化。我已经把每个文件里边的父节点和多个属性的节点生成了一个个的类。不过这个工程对个人研究来说工作量也是相当巨大。只能一点点的来。

yangshengquan1026, 2024-12-06:
andlaus, thank you for your prompt reply! Yes, you are correct. My dop-snref will be queryable if it is defined in this file's query, but not if it is defined in another file. This is because the tool used by Softing to generate the test.pdx file works correctly, but the tool used by ODX Tools does not. I have tried modifying the old version of ODX Tools to fix this issue, but it worked only temporarily. It seems that ODX Tools stores all the fields of each file in a common dictionary, such as all the fields of diagservice in one dictionary, and all the fields of dop in another. I am not very familiar with Python, but I am trying to understand the underlying principles so that I can convert it to C/C++ or Rust. I have already generated classes for each file's parent node and multiple attribute nodes. However, this project is still quite large for personal research, and I will have to work on it gradually.

andlaus, 2024-12-06:
> My dop-snref will be queryable if it is defined in this file's query, but not if it is defined in another file

This might be a vagueness in the ODX standard: I believe that SNREFs must be resolvable in all contexts that occur in a given database (i.e., if there is a SNREF in an ECU-SHARED-DATA layer, it must be resolvable with just the data provided by that layer). Since -- to my knowledge -- the specification makes no explicit statement about this, the softing tools might take a different approach (if they are heavily based on XPATH expressions, on-the-fly lookups might even be easier for them to implement). As said, using ODXLINK references (DOP-REF) avoids this issue...

cmydd, 2024-12-07:
这个问题没有修复.SNREF必须提取到ecuvariant里面才不会出错

yangshengquan1026, 2024-12-07:
> > 如果我的 dop-snref 在这个文件中的查询定义，将会是可查询的，但如果它在另一个文件中定义，则不可查询
> 
> 这可能是 ODX 标准的一个模糊节点：我认为 SNREF 必须在给定数据库中出现的所有上下文中劳动力解析（即，如果 ECU-SHARED-DATA 层中存在 SNREF，则必须仅使用该层中可用同样，由于规范对此没有明确说明，因此软化工具可能会采用不同的方法（如果它们主要基于 XPATH 表达式，则立即查找对它们来说甚至可能更容易实现）。如前所述，使用 ODXLINK引用（DOP-REF）可以避免此问题……

thank you！

yangshengquan1026, 2024-12-07:
> > My dop-snref will be queryable if it is defined in this file's query, but not if it is defined in another file
> 
> This might be a vagueness in the ODX standard: I believe that SNREFs must be resolvable in all contexts that occur in a given database (i.e., if there is a SNREF in an ECU-SHARED-DATA layer, it must be resolvable with just the data available in that layer). Since -- to my knowledge -- the specification makes no explicit statement about this, the softing tools might take a different approach (if they are heavily based on XPATH expressions, on-the-fly lookups might even be easier for them to implement). As said, using ODXLINK references (DOP-REF) avoids this issue...

thank you！

yangshengquan1026, 2024-12-07:
> 这个问题没有修复.SNREF必须提取到ecuvariant里面才不会出错 但是c++ 很好实现 python都是引用 动一个其他全部会改变

有没有想法像他们这样搭建一个c++/rust版本的odxtools?

cmydd, 2024-12-07:
c++实现完成了 他这个基本上是没问题的  只是snref会有问题 之前有做过  但是没有按照标准的iso 22901来实现. 现在在测试  有一些问题

yangshengquan1026, 2024-12-07:
> c++实现完成了 他这个基本上是没问题的 只是snref会有问题 之前有做过 但是没有按照标准的iso 22901来实现. 现在在测试 有一些问题

我看这个odxtools和那个java的都有了，但是现在就c++和rust的还比较欠缺。

cmydd, 2024-12-12:
> > c++实现完成了 他这个基本上是没问题的 只是snref会有问题 之前有做过 但是没有按照标准的iso 22901来实现. 现在在测试 有一些问题
> 
> 我看这个odxtools和那个java的都有了，但是现在就c++和rust的还比较欠缺。

请问 java的哪里有开源的呢

andlaus, 2025-01-16:
while not a complete fix, #373 should help you to get around the SNREF issue by retargeting all SNREFS to your ECU of interest. I'll thus close the issue for now...

closed 2025-01-16 by andlaus