mercedes-benz/odxtools #375: a bug with lengthleylarameter

cmydd, 2024-12-16:
hello
a odx like this

PARAM xsi:type="LENGTH-KEY" SEMANTIC="DATA" ID="Req_ReadMemorByAddre.Param_LengtFormaIdent">
        SHORT-NAME>Param_LengtFormaIdent</SHORT-NAME>
        LONG-NAME>Length Format Identifier</LONG-NAME>
        BYTE-POSITION>1</BYTE-POSITION>
        BIT-POSITION>4</BIT-POSITION>
        DOP-REF DOCREF="ES_DataLibra" DOCTYPE="LAYER" ID-REF="DOP_UINTDEC4Bits8x1F"/>
       /PARAM>
       PARAM xsi:type="LENGTH-KEY" SEMANTIC="DATA" ID="Req_ReadMemorByAddre.Param_AddreFormaIdent">
        SHORT-NAME>Param_AddreFormaIdent</SHORT-NAME>
        LONG-NAME>Address Format Identifier</LONG-NAME>
        BYTE-POSITION>1</BYTE-POSITION>
        DOP-REF DOCREF="ES_DataLibra" DOCTYPE="LAYER" ID-REF="DOP_UINTDEC4Bits8x1F"/>
       /PARAM>

 DATA-OBJECT-PROP ID="DOP_UINTDEC4Bits8x1F">
        DIAG-CODED-TYPE xsi:type="STANDARD-LENGTH-TYPE" BASE-DATA-TYPE="A_UINT32">
        BIT-LENGTH>4</BIT-LENGTH>
        BIT-MASK>0F</BIT-MASK>
       /DIAG-CODED-TYPE>

lengthkeyparameter.py
line 105
# emplace a value of zero into the encode state, but pretend the bits not to be used
        n = odxrequire(self.dop.get_static_bit_length()) + encode_state.cursor_bit_position
        tmp_val = b'\x00' * ((n + 7) // 8)
        encode_state.emplace_bytes(tmp_val, obj_used_mask=tmp_val)

the encode_state.cursor_bit_position is 4 and self.dop.get_static_bit_length() is 4
if encode_state.cursor_bit_position != 0:
     encode_state.cursor_bit_position = (odxrequire(self.dop.get_static_bit_length()) + encode_state.cursor_bit_position) % 8

if not like this.encodestate will throw exception
encodestate.py
line 191
        if self.cursor_bit_position != 0:
            odxraise("EncodeState.emplace_bytes can only be called "
                     "for a bit position of 0!", RuntimeError)

andlaus, 2024-12-16:
thanks for the catch. (seems like nobody stumbled over non-byte aligned lentgh keys so far.) The fix is to add
```python
        encode_state.cursor_bit_position = 0
```
before `emplace_bytes()` is called for the placeholder value. can you make a PR?

cmydd, 2024-12-16:
> thanks for the catch. (seems like nobody stumbled over non-byte aligned lentgh keys so far.) The fix is to add
> 
> ```python
>         encode_state.cursor_bit_position = 0
> ```
> 
> before `emplace_bytes()` is called for the placeholder value. can you make a PR?
sorry i dont knwo hao to make a pr.i can give you the odx file. search length-key or "Req_ReadMemorByAddre", you can find a lot of them.

cmydd, 2024-12-16:
[odxfile.zip](https://github.com/user-attachments/files/18149592/odxfile.zip)

closed 2024-12-17 by andlaus