mercedes-benz/odxtools #377: about PARAM-SNREF

cmydd, 2024-12-20:
e.g
PARAM xsi:type="VALUE" SEMANTIC="DATA">
        SHORT-NAME>Param_DTCValue</SHORT-NAME>
        LONG-NAME>DTC Value</LONG-NAME>
        BYTE-POSITION>2</BYTE-POSITION>
        DOP-REF DOCREF="ES_DataLibra" DOCTYPE="LAYER" ID-REF="DOP_UINTHEX3Bytes"/>
       /PARAM>

ENV-DATA-DESC ID="ENVDESC_DTCExtenDataRecor">
       SHORT-NAME>ENVDESC_DTCExtenDataRecor</SHORT-NAME>
       LONG-NAME>DTC Extended Data Record</LONG-NAME>
       PARAM-SNREF SHORT-NAME="Param_DTCValue"/>
       ENV-DATAS>

environmentdatadescription.py
line 133
if not isinstance(prev_dop, DtcDop):
DOP_UINTHEX3Bytes is touble_code.the dop is not DtcDop.

andlaus, 2024-12-20:
I read up the spec, and it indeed allows any parameter that uses a DOP with a A_UINT32 base data type while odxtools currently only allows DTC-DOPs. I'll only be able to fix this in January, though...

> if if self.termination == "END-OF-PDU",termination_sequence is None. An exception is thrown

Hm, I cannot see how [`MinMaxLenthType.__termination_sequence()`](https://github.com/mercedes-benz/odxtools/blob/main/odxtools/minmaxlengthtype.py#L57-L72) can ever return `None`...

cmydd, 2024-12-21:
def __termination_sequence(self) -> bytes:
        """Returns the termination byte sequence if it isn't defined."""
        # The termination sequence is actually not specified by ASAM
        # for A_BYTEFIELD but I assume it is only one byte.
        termination_sequence = b''
        if self.termination == "ZERO":
            if self.base_data_type not in [DataType.A_UNICODE2STRING]:
                termination_sequence = bytes([0x0])
            else:
                termination_sequence = bytes([0x0, 0x0])
        elif self.termination == "HEX-FF":
            if self.base_data_type not in [DataType.A_UNICODE2STRING]:
                termination_sequence = bytes([0xFF])
            else:
                termination_sequence = bytes([0xFF, 0xFF])
        return termination_sequence

if self.termination is "END-OF-PDU",__termination_sequence return None

e.g
 DATA-OBJECT-PROP ID="DOP_BYTEFIELD0ToEOPBytes">
       SHORT-NAME>DOP_BYTEFIELD0ToEOPBytes</SHORT-NAME>
       LONG-NAME>BYTE FIELD 0 to EOP Bytes</LONG-NAME>
       DESC>
        p>Bytefield beliebiger Länge (incl. 0 Bytes)</p>
       /DESC>
       COMPU-METHOD>
        CATEGORY>IDENTICAL</CATEGORY>
       /COMPU-METHOD>
       DIAG-CODED-TYPE xsi:type="MIN-MAX-LENGTH-TYPE" BASE-DATA-TYPE="A_BYTEFIELD" TERMINATION="END-OF-PDU">
        MIN-LENGTH>0</MIN-LENGTH>
       /DIAG-CODED-TYPE>
       PHYSICAL-TYPE BASE-DATA-TYPE="A_BYTEFIELD"/>
      /DATA-OBJECT-PROP>

cmydd, 2024-12-21:
the is another issue
diaglayer.py
line 408
                if not gnr_found:
                    raise e

if i have 3 service,One of them meets the conditions. It shouldn't be throwing an exception.

               if not gnr_found:
                   continue
Whether it's correct or not

andlaus, 2024-12-21:
> if i have 3 service,One of them meets the conditions. It shouldn't be throwing an exception. [...] Whether it's correct or not

in this case, the telegram which you are trying to decode could not be interpreted by odxtools (neither as a request, nor as a response or as a global negative response). In my opinion, raising a `DecodeError` is well justified in this case. Most likely your ODX dataset is incomplete (i.e., does not describe all data which the ECU can send) or you are stumbling over some issues encountered during the decoding procedure and you are not using the [non-strict mode](https://github.com/mercedes-benz/odxtools?tab=readme-ov-file#using-the-non-strict-mode).

andlaus, 2024-12-21:
(as a work-around you can catch the exception in the code that calls `.decode`...)

cmydd, 2024-12-23:
thanks, I understand

andlaus, 2025-01-14:
https://github.com/mercedes-benz/odxtools/pull/379 should fix this. would be great if tested it with your stuff...

andlaus, 2025-01-16:
with #379 merged, it seems like this issue has been resolved and can be closed. If this is not the case, please elaborate what goes wrong and re-open it...

closed 2025-01-16 by andlaus