mercedes-benz/odxtools #421: Fix object lifetime management

andlaus, 2025-04-29:
The odxtools database object currently seem to mess up python's garbage collector: when running
```python
import odxtools
import time

print("loading database", flush=True)
db = odxtools.load_file("my_big_pdx_file.pdx")
print("database loaded", flush=True)
time.sleep(20)
print("deleting database", flush=True)
del db
print("database deleted", flush=True)
time.sleep(20)
print("done", flush=True)
```

the memory used by the `db` object is not freed after the "database deleted" string is printed but only at program exit, i.e., after "done" appears.

I suspect that this is due to python's garbage collector getting confused by cyclic references, but unfortunately I have not found a good way to debug this, and my limited experiments with weak references from python's `weakref` module also did not fix the issue. @kayoub5, @nada-ben-ali, @zariiii9003: any ideas how to get to the bottom of this?

kayoub5, 2025-04-29:
Those are common mis-conception about garbage collectors:
* Garbage collocation does not run when you are done with the object, quite the opposite, most GCs (I have seen) run when the program ask to allocate more memory, more on why below.
* Some GCs have "scope exit" logic, basically GC auto run on local objects of functions for example, since are those are easier to detect and clean,  your code have global module variables, GC typically don't aggressively run until module logic is over.
* GCs usually require putting the entire application/thread in a pause state in order to run, so typical imlimentations avoid interrupting your code as much as possible (usually wait for a sleep/ io operation), unless RAM is very limited.
* A process does not return the memory it longer needs to the os, the reason is that "it may need it again soon", and requesting more RAM from os is slow (in some cases, running a GC is faster than asking os for RAM). so process just mark RAM as "os free to take it back if it really needs to, but I prefer to keep it if possible".

andlaus, 2025-04-30:
Isn't python's main memory reference counting? I'm not sure how exactly it is implemented in cpython, but the other refcounting implementation I have experience with (C++'s smart pointers) frees memory of objects which's reference count reaches zero immediately. (This is also a very fast operation...)

> lambda functions keep references to objects simply because the lambda code uses them, which makes them hard to notice.

sure, but as soon as the refcount of the lambda function reaches zero, it should be deleted...

kayoub5, 2025-04-30:
A memory freed does not mean the process memory goes down immediately, a process always keep the memory pages for fast re-allocation.

andlaus, 2025-04-30:
> A memory freed does not mean the process memory goes down immediately, a process always keep the memory pages for fast re-allocation.

sure, but I assume that the heap allocators will free memory eventually (in my case "eventually" means after deleting a > 5 GB `Database` object.)

andlaus, 2025-04-30:
I would be surprised if there was a formal "out of band" mechanism of the Linux kernel to signal to userspace that there is memory pressure...

kayoub5, 2025-05-02:
Since you confirmed that calling `gc.collect()` free the memory, this confirms that it's just the GC being lazy (which is a known behavior in all GC implementations regardless of language).

GC does not run just because the ref count of a single object became zero.

andlaus, 2025-05-05:
as far as I'm aware, in contrast to full garbage collection, reference counting does not cause any significant delay between the time an object is deleted and when it's memory freed. Unfortunately, it does not work if there are cycles and sadly for us the ODX data model definitely features cycles as soon as references are resolved. This hypothesis is evidenced by the fact that if `Database.refresh()` is replaced by a no-op function, the memory occupied by `Database` objects gets freed immediately. (`del db` still takes a bit of time, but it is approximately an order of magnitude quicker than the garbage collector.)

My point with this issue is that making `Database` work nicely with reference counting would be great, but I found it pretty difficult to debug, so if somebody who's smarter than me has an idea of how to accomplish this, please report here. An idea would be to use weak references to any resolved ODXLINK or SNREF links. (Unfortunately, I ran into some trouble when trying this approach, though...)

closed 2025-09-24 by andlaus