mercedes-benz/odxtools #429: PDX File Read Error

parth-20-07, 2025-05-20:
Am trying to use the cli python version to read a .pdx file but am getting following error:

```
❯ python3 -m odxtools list -a "PDX_FLL_Appl_PBL_SBL_32327978AO_32294403AB_32294404AB_20250414.pdx"
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/__main__.py", line 6, in <module>
    _main()
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/__init__.py", line 80, in _main
    _main.start_cli()
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/cli/main.py", line 67, in start_cli
    tool.run(args)
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/cli/list.py", line 215, in run
    odxdb = _parser_utils.load_file(args)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/cli/_parser_utils.py", line 35, in load_file
    return _load_file(pdx_file_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/loadfile.py", line 25, in load_file
    return load_pdx_file(str(file_name))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/loadfile.py", line 11, in load_pdx_file
    db.refresh()
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/database.py", line 143, in refresh
    subset._resolve_odxlinks(self._odxlinks)
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/comparamsubset.py", line 80, in _resolve_odxlinks
    comparam._resolve_odxlinks(odxlinks)
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/comparam.py", line 47, in _resolve_odxlinks
    self._dop = odxlinks.resolve(self.dop_ref, DataObjectProperty)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/odxlink.py", line 212, in resolve
    odxraise(
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/exceptions.py", line 50, in odxraise
    raise error_type(message)
KeyError: "ODXLINK reference OdxLinkRef('id') could not be resolved for any of the document fragments (OdxDocFragment(doc_name='ISO_13400_2_DIS_2015', doc_type=<DocType.COMPARAM_SUBSET: 'COMPARAM-SUBSET'>),)"
```

andlaus, 2025-05-21:
KeyError: "ODXLINK reference OdxLinkRef('id') could not be resolved for any of the document fragments (OdxDocFragment(doc_name='ISO_13400_2_DIS_2015', doc_type=<DocType.COMPARAM_SUBSET: 'COMPARAM-SUBSET'>),)"

it seems like your PDX file is not correct: it references the data object property "id" but this is not defined in the communication parameter subset `ISO_13400_2_DIS_2015`. you can try the non-strict mode:
```
python3 -m odxtools --no-strict list -a "PDX_FLL_Appl_PBL_SBL_32327978AO_32294403AB_32294404AB_20250414.pdx"
```

parth-20-07, 2025-05-21:
I recieved another error post that:

```
❯ python3 -m odxtools --no-strict list -a "PDX_FLL_Appl_PBL_SBL_32327978AO_32294403AB_32294404AB_20250414.pdx"
ODXLINK reference OdxLinkRef('id') could not be resolved for any of the document fragments (OdxDocFragment(doc_name='ISO_13400_2_DIS_2015', doc_type=<DocType.COMPARAM_SUBSET: 'COMPARAM-SUBSET'>),)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/__main__.py", line 6, in <module>
    _main()
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/__init__.py", line 80, in _main
    _main.start_cli()
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/cli/main.py", line 67, in start_cli
    tool.run(args)
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/cli/list.py", line 215, in run
    odxdb = _parser_utils.load_file(args)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/cli/_parser_utils.py", line 35, in load_file
    return _load_file(pdx_file_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/loadfile.py", line 25, in load_file
    return load_pdx_file(str(file_name))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/loadfile.py", line 11, in load_pdx_file
    db.refresh()
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/database.py", line 143, in refresh
    subset._resolve_odxlinks(self._odxlinks)
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/comparamsubset.py", line 80, in _resolve_odxlinks
    comparam._resolve_odxlinks(odxlinks)
  File "/home/parth/.local/lib/python3.11/site-packages/odxtools/comparam.py", line 48, in _resolve_odxlinks
    self._physical_default_value = self._dop.physical_type.base_data_type.from_string(
                                   ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'physical_type'
```

andlaus, 2025-05-24:
this is a follow-up error which is the result of the reference to the DOP pointing to the void. I'm afraid, that the fastest way of getting around this is fixing your dataset (possibly, it is sufficient to add `DOCREF` and `DOCTYPE` to the `DOP-REF` pointing to `id` in the `ISO_13400_2_DIS_2015` COMPARAM-SUBSET)

andlaus, 2025-06-06:
I don't think that something should be done about this issue on the odxtools level. If you disagree, feel free to open a pull request with a fix. closing.

closed 2025-06-06 by andlaus